tock/tockloader #1: tockloader is annoying about argument order

ppannuto, 2016-12-23:
```bash
$ tockloader listen --port /dev/tty.usbserial-000011
usage: tockloader [-h] [--port PORT] {flash,listen} ...
tockloader: error: unrecognized arguments: --port /dev/tty.usbserial-000011
$ tockloader --port /dev/tty.usbserial-000011 listen
Hello World!
```

bradjc, 2016-12-23:
Is this fixable? I'm just using python's argparse.

ppannuto, 2016-12-23:
Dunno. I was just about to leave but wanted to make sure the hail I grabbed worked before I left and ran into this. Feels like we can't be the only ones annoyed by this.

I can look into it later. Also would love to figure out where that 00011 is coming from (this Hail is nominally `:06` according to its sticker)

commit 9294b5d3 (2017-01-09): Allow global arguments in any order

Closes #1.
closed 2017-01-09 by ppannuto