tock/tockloader #2: Macports: Python 3.6: tockloader executable not created by pip?

cxbrooks, 2016-12-29:
When I installed tockloader under Mac Ports, it seems like the tockloader binary was not created?

My installation was:

```
sudo port install python36-pip
sudo port install py36-pip
```
However, there was no tockloader binary installed?

I ended up doing:

```
sudo ln -s /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tockloader/main.py /opt/local/bin/tockloader
sudo chmod a+x /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tockloader/main.py
```

Below is what happens when I try again to install tockloader, which looks OK to me:
```
bash-3.2$ sudo -H pip-3.6 install tockloader
Requirement already satisfied: tockloader in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages
Requirement already satisfied: pyserial>=3.0.1 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from tockloader)
Requirement already satisfied: crcmod>=1.7 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from tockloader)
```

The directory contents are below:

```
bash-3.2$ ls /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages
README                                  easy_install.py                         setuptools
README.txt                              pip                                     setuptools-32.3.0-py3.6.egg-info
__pycache__                             pip-9.0.1-py3.6.egg-info                tockloader
crcmod                                  pkg_resources                           tockloader-0.1.0.dist-info
crcmod-1.7-py3.6.egg-info               pyserial-3.2.1.dist-info                tockloader-0.2.0-py3.6.egg
easy-install.pth                        serial
bash-3.2$ ls /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tockloader
__init__.py     __pycache__     main.py
```

Here's the link I created:

```
bash-3.2$ ls -l /opt/local/bin/tockloader 
lrwxr-xr-x  1 root  admin  106 Dec 28 19:17 /opt/local/bin/tockloader -> /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tockloader/main.py
```

This might be a Mac Port-specific problem, I could retry with Brew.

brghena, 2017-01-04:
I tried this today on an uninitialized macbook using Brew.

```
brew install python3
sudo pip3 install tockloader
tockloader --help
```

Everything worked fine. That being said, I've certainly had strange errors with pip in the past.

cxbrooks, 2017-01-04:
I'm guessing this is a Macport issue and my unfamiliarity with Python.

It looks like this script is created: /opt/local/Library/Frameworks/Python.framework/Versions/3.6/bin/tockloader

It could be that I missed a step about adding that directory to my path.

This issue can probably be closed, it is really a problem with Python (and me :-) )

closed 2017-01-04 by cxbrooks