tock/tockloader #4: Passing multiple apps to `flash` does not align them

ppannuto, 2017-01-24:
```
$ tockloader flash build/cortex-m4/app.bin ../../sensors/build/cortex-m4/app.bin
...
$ tockloader list
No serial port specified. Discovering attached serial devices...
Using "/dev/cu.usbserial-003032 - Hail IoT Module - TockOS - Hail IoT Module - TockOS"
[App 0]
  Name:                  printf_long
  Total Size in Flash:   4096 bytes

[App 1]
  Name:                  sensors
  Total Size in Flash:   8192 bytes
  [WARNING] App is misaligned for the MPU
```

bradjc, 2017-01-25:
The reason behind this is that the `flash` command was taken from `sload` and is also used to flash the kernel, so it doesn't make the assumption that it can introspect the binaries being passed to it. My take would be to add a new verb (not sure what would be good) that is only used to flash apps and ensures that only the apps being flashed at that time end up on the board (to separate the command from `add`).

alevy, 2017-01-25:
since there is `add` and `remove`, how about `replace`, which means replace all current apps with the apps i'm giving you?

bradjc, 2017-01-25:
`replace` already exists, and replaces the app of the same name. It's
designed for debugging when you want to test multiple apps but only iterate
one.

ppannuto, 2017-01-25:
Is there an `update` command? The current `replace` sounds more like
`update` to me

bradjc, 2017-01-25:
There is no update currently.

What I don't like about using replace for the original problem is that its
not clear when you are using tockloader for the first time on a new board
that you have anything to replace.

ppannuto, 2017-01-26:
Maybe we don't need a separate command at all? We already have code in tockloader that parses the tbf, which'll fail if it isn't an app. The first few bytes of a kernel are pretty identifiable (starts with 0x20xxxxxx [this addr specified by arm, should be chip-independent], then at least a dozen words with the last bit set to 1).

Tockloader could just figure out whether its an app or kernel that's been passed and do the right thing?

alevy, 2017-01-26:
it could. i'm personally a bit weary of too much magic, definitely when it's not built on top of non-magic building blocks (e.g. 'program-kernel' and 'program-app' commands, and 'program' that just tries to guess and call to one of them). But meh...

ppannuto, 2017-01-26:
At the same time, I've already tried to flash an elf on accident (current
behavior silently drops it), so a little sanity checking of what's being
flashed and helpful error/warning messages if they aren't what's expected
seems like a good idea to me. Feels like once that's there, then tockloader
has confidence in what it's flashing.

alevy, 2017-01-26:
yeah. i guess it's a question of how closely we want to tie `tockloader` to Tock, but there is nothing inherently illegal in flashing in place of the kernel something that happens to look like an app or an ELF, is there? In any case, _I_ don't feel strongly about this, just my two cents.

ppannuto, 2017-01-26:
Inherently illegal, no, likely wrong / not what you wanted, yes. A flag
escape (like the compile-flag check) is probably a fine compromise here.

More generally, I don't mind tying *tock*loader to Tock, but that may just
be me :)

bradjc, 2017-02-24:
Added `tockloader install`.

closed 2017-02-24 by bradjc