tock/tockloader #10: Feature Request: Signal `tockloader listen` processes

bradjc, 2017-05-27:
In my experience any tockloader command will fail if another terminal has `tockloader listen` running. Therefore it would be nice if on any command, tockloader would signal all other tockloader processes, and if those were `tockloader listen` on the same port, they would release the serial port, wait for the signalling process to finish, and then resume listen. This would make tockloader work seamlessly with itself.

Note: I'm not actually sure what the best way to get tockloader processes to communicate is.

alevy, 2017-05-28:
pid file? somewhere in the user's home directory? perhaps named by the serial port as well so it only interrupts processes referring to the same serial port?

ppannuto, 2017-06-05:
Okay, so I spent entirely too long today trying to get this to work (headway in `listen-ipc` branch), and ended up failing.

The biggest reason I couldn't get this to work is that miniterm isn't designed to be killed by another thread. It's reads are blocking until there's some form of I/O and there's no way to cancel them prematurely without somehow injecting characters into stdin.

The next route I tried to go down was just killing off the process entirely, first by using `execvp`, but that meant that miniterm's atexit handlers didn't run and your terminal was corrupt. Next I tried some hackery around `spawnvp`ing a fresh copy and completely killing off the original process, but that didn't work either.

Despite sounding simple, this feature is proving non-trivial

ppannuto, 2017-06-06:
Aha! Sorry to use github issue as a rubber duck, but the right combination seems to be manually calling miniterm's atexit handlers and then execvp'ing - seems to be working now, just need to clean it up

commit 757f38a1 (2017-06-06): add inter-process coordination

The approach we take is to have every tockloader instance either
connect to a server named uniquely for each serial port, or if one
doesn't exist make that server

For most operations, the server will just reject other tockloader
instances. If the running process is a `listen`, however, we want
to essentially pause the listen while the other operation goes through
and then resume again. Unfortunately, due the the way that miniterm
is architected, this proves much easier to do by killing off the
current listen process and spinning up a new one, with a bonus flag
 [... cut]
commit afb4ed11 (2017-06-08): add inter-process coordination

The approach we take is to have every tockloader instance either
connect to a server named uniquely for each serial port, or if one
doesn't exist make that server

For most operations, the server will just reject other tockloader
instances. If the running process is a `listen`, however, we want
to essentially pause the listen while the other operation goes through
and then resume again. Unfortunately, due the the way that miniterm
is architected, this proves much easier to do by killing off the
current listen process and spinning up a new one, with a bonus flag
 [... cut]
PR #12 (2017-06-09): add inter-process coordination
The approach we take is to have every tockloader instance either
connect to a server named uniquely for each serial port, or if one
doesn't exist make that server

For most operations, the server will just reject other tockloader
instances. If the running process is a `listen`, however, we want
to essentially pause the listen while the other operation goes through
and then resume again. Unfortunately, due the the way that miniterm
is architected, this proves much easier to do by killing off the
current listen process and spinning up a new one, with a bonus flag
t [... cut]
closed 2017-06-09 by bradjc