tock/tockloader #19: OpenOCD support for NRF

lemmi, 2018-01-23:
Hi,

with version 0.10.0, OpenOCD gained a couple of improvements for the NRF chips. I wrote a small script that is able to flash kernel and app bundles. Now I'm interested in adding support for openocd to tockloader, since openocd seems to be much more available than the jlink and it has support for different programmers aswell. Flashing an NRF via stlink is possible aswell.

So far my understanding it that I need to implement the [BoardInterface](/helena-project/tockloader/blob/master/tockloader/board_interface.py) for openocd. Is there anything else I need to take into consideration?

cpluss, 2018-03-10:
@bradjc this would be nice, since openocd can be used to flash launchxl board as well (I think, still need to verify)

alevy, 2018-03-10:
I agree that openocd would be nicer to support over JLink. It's easier to get openocd (at least on Linux)

bradjc, 2018-03-10:
I'm all for adding openocd support and yes, implementing BoardInterface is
all that should be required. I'm not really sure how to use openocd and
what commands are required, but I can help with integrating the support
into tockloader.

>

lemmi, 2018-03-10:
Ok, I'll dig out the script I use to flash the nrf51 and start on a prototype.

bradjc, 2018-03-10:
I pushed a few commits that should make this easier. Really just need to implement `flash_binary`, `read_range`, `erase_page`, and `determine_current_board` for openocd; the other commands have default implementations.

We will need an `--openocd` flag to tell tockloader to use that channel. We may also need `--openocd-device` (not sure) and if so we may want to un-deprecate `--jtag-device` so that it can be used for any of the JTAG channels. Maybe we want an even more generic flag to specify the device, I'm not sure.

closed 2018-06-12 by bradjc