tock/tockloader #50: nrf52840dk different tockloader listen behavior

hudson-ayers, 2020-01-10:
For most boards in Tock, you pass the same commands to `tockloader listen` as you do to most other tockloader commands. For the nrf52840dk, however, this is not the case.

To `install`, or `erase-apps`, or `list` on the nrf52840dk, you use

`tockloader [command] --board nrf52dk --jlink`

However in order to `listen`, you must use just

`tockloader listen --board nrf52dk` and then press the reset button.

There are two issues here
1. For the nrf52840dk, `tockloader listen` does not reset the board like it does for Hail or Imix, so if you have an app that only prints near startup you will not see any output without resetting the board
2. For the nrf52840dk, all commands require `--jlink` except `listen`, where if you pass `--jlink` then you will not receive any console output at all, but also will not receive any error.

bradjc, 2020-01-10:
Tockloader does not intend to reset the board when running listen, but as far as I know it is a limitation with the OS-level UART drivers that user level code can't control how pins are toggled well enough, causing an undesirable reset on boards like Hail and imix where the FTDI chip is connected to the reset pin. On the nRF the UART HW in the jtag programmer is not connected to reset and therefore `tockloader listen` doesn't reset the board.

`tockloader listen --jlink` is in fact very different as it uses the Segger RTT protocol to communicate with the board.

hudson-ayers, 2020-01-10:
Gotcha, so this is not a bug after all. However it is worth noting that `tockloader listen --jlink` does not work for the nrf52840dk despite the board having onboard integrated JTAG and the `--jlink` option being required for all other tockloader commands on this board.

bradjc, 2020-01-10:
Yes that is odd now that you point it out. On one hand the way it is makes sense, since no argument means use UART, which is normally what we want, and specifying --jlink means use segger jlink, which is what the RTT needs.

However, that requires really knowing what is going on, and I think it probably makes more sense to  add a new option to specify you want the RTT version of listen.

bradjc, 2020-03-12:
Fixed in master. RTT now uses the `--rtt` flag.

closed 2020-03-12 by bradjc