tock/tockloader #58: New release on pip3?

gendx, 2020-04-14:
[OpenSK](https://github.com/google/OpenSK)'s deploy script depends on the `App.get_binary()` function from tockloader.

https://github.com/tock/tockloader/blob/6e48d5f79585ccb622aff5d0fca4ef54ce4ce94f/tockloader/app.py#L95-L111

However, this function is not available in the latest released version of tockloader on pip3. This causes some failures: https://github.com/google/OpenSK/issues/79.

Could you release a new version of tockloader on pip3?

gendx, 2020-04-15:
cc @bradjc

bradjc, 2020-04-15:
I do want to make a new release. Are you running from master? Have you encountered any issues?

jmichelp, 2020-04-15:
Yes, I've been using master for a while (that's why I wasn't picking up that some of the methods were not in the pypi release). No issue so far for OS X and Linux on Nordic boards.

gendx, 2020-04-20:
Any update on this @bradjc?

gendx, 2020-04-28:
Released: https://pypi.org/project/tockloader/#history

closed 2020-04-28 by gendx