tock/tockloader #64: Dropped Messages

kevtan, 2020-11-24:
It seems like `tockloader listen` drops the first messages that are printed using the `debug!` macro. Here is the context:

- I flash the stock kernel onto the nano 33.
- I flash the `c_hello` program onto the nano33.
- I execute `tockloader listen`.
- I expect to see `"Initialization complete. Entering main loop."` printed.
- I do not see this.

As a remedy suggested by Hudson, I added a bunch of copies of the `debug!` macro (shown below) and am able to—non-deterministically—see some of the last statements.

[image]

Here are some example runs:

```text
❯ tockloader listen
[INFO   ] No device name specified. Using default name "tock".
[INFO   ] Using "/dev/cu.usbmodem14301 - Nano 33 BLE _ TockOS".
[INFO   ] Listening for serial output.
omplete. Entering main loop.
Initialization complete. Entering main loop.
Initialization complete. Entering main loop.
Initialization complete. Entering main loop.
Initialization complete. Entering main loop.
Hello World!
```

```text
❯ tockloader listen
[INFO   ] No device name specified. Using default name "tock".
[INFO   ] Using "/dev/cu.usbmodem14301 - Nano 33 BLE _ TockOS".
[INFO   ] Listening for serial output.
main loop.
Initialization complete. Entering main loop.
Initialization complete. Entering main loop.
Initialization complete. Entering main loop.
Hello World!
```

I'm using this duplication strategy right now because I don't have the bandwidth to debug this, but I've made an issue to document the bug.

hudson-ayers, 2020-11-24:
This works on bleeding edge tockloader, just not in the latest released version.

To get the fix, clone this repo, then run `python3 setup.py build` and `python3 setup.py install`, then try again and it should work correctly.

kevtan, 2020-11-25:
I built tockloader from source and there is still issues with dropped messages. Admittedly, it got MUCH better. In particular, I am still missing my first `debug!` statement. For instance, if I have 2 of these, then I see 1 in the output. If I have only 1 of these, then I see 0 in the output.

kevtan, 2020-11-28:
Update: I think the issue with dropped messages has not been fixed even on the bleeding edge of tockloader. In the following code block, I query the version of tockloader that I'm using:
```
❯ python3 -m tockloader.main --version
1.6.0-dev
```
And then, I run tockloader listen a few times:
```
❯ python3 -m tockloader.main listen
[INFO   ] No device name specified. Using default name "tock".
[INFO   ] Using "/dev/cu.usbmodem14301 - Nano 33 BLE - TockOS".
[INFO   ] Listening for serial output.
omplete. Entering main loop.
Initialization complete. Entering main loop.
Hello World!

[INFO   ] No device name specified. Using default name "tock".
[INFO   ] Using "/dev/cu.usbmodem14301 - Nano 33 BLE - TockOS".
[INFO   ] Listening for serial output.

[INFO   ] No device name specified. Using default name "tock".
[INFO   ] Using "/dev/cu.usbmodem14301 - Nano 33 BLE - TockOS".
[INFO   ] Listening for serial output.
Hello World!
```

hudson-ayers, 2020-11-30:
Can you post your OS (including version) here? This seems to be platform specific.

kevtan, 2020-11-30:
[image]

bradjc, 2021-08-09:
I _hope_ that https://github.com/tock/tock/pull/2731 has finally fixed this.

closed 2021-08-09 by bradjc