tock/tockloader #67: Flashing apps via openocd error

alexandruradovici, 2021-04-19:
I am using tockloader --openocd to upload apps to the microbit v2. I have a very strange behavior:

- writing the first app works
- writing a second app will erase all apps

Using the tockloader with the serial bootloader seems to work fine.

bradjc, 2021-04-19:
What happens if you use openocd with `--bundle-apps`? I assume it will work. I'm also guessing you were flashing small apps. This problem seems to appear on some boards, where `program` erases flash that is not being written, which is at a lower address than what is being flashed, and that happens to erase the first app, which then breaks the linked-list of apps. I don't know why implementations of the `program` command in openocd do not protect against this.

One workaround is to set a minimum size of apps using the `BOARD_APP_DETAILS` static variable:

```python
	BOARDS_APP_DETAILS = {
	    'default': {'order': 'size_descending',
	                'size_constraint': 'powers_of_two',
	                'size_minimum': 0},
	    'nrf52dk': {'size_minimum': 4096},
	    'edu-ciaa': {'cmd_flags': {'bundle_apps': True,
	                               'openocd': True}},
	    'arty': {'size_minimum': 0x10000,
	             'size_constraint': None},
	}
```

This has the effect of forcing apps to align to erase sizes, and works around the issue with `program`. Another workaround is to always bundle apps, so all apps are only written once and all together.

brghena, 2021-08-06:
I can confirm that adding a `BOARDS_APP_DETAILS` entry for the `microbit_v2` with `bundle_apps` or with `size_minimum` of 4096 resolves the issue. Is there a preference as to which should be used?

hudson-ayers, 2021-08-06:
I would advocate for `bundle-apps`

bradjc, 2021-08-06:
As I think about this, I don't really like the bundle-apps feature. That is more of a hack, and is pretty opposed to the normal tockloader spirit of treating tock apps as "normal" applications (i.e. you wouldn't delete and re-install every app on your phone to add a new one).

The size minimum is a much more straightforward fix.

hudson-ayers, 2021-08-06:
Philosophically I see that argument, but in practical terms I see the tradeoffs as follows:

- `min_size=4096` --> limits the number of apps that can be loaded
- `bundle_apps` --> makes adding or removing apps take a little longer (an otherwise quick process on most boards, though I have not used the microbit)

I consider the former worse than the latter

hudson-ayers, 2021-08-06:
My understanding is that libtock-rs is going to strive to support much smaller apps post-2.0. I agree that the minimum app size isn't a huge deal, I just think that a negligibly larger time to flash apps is even less of a big deal, given that Branden said it wasn't a noticeable amount of time anyway. That said, I haven't tried it myself, and maybe it is in fact noticeable once you have several large apps.

bradjc, 2021-08-06:
With jtag the writing time will probably always be negligible. But the microbit also supports the serial bootloader, and that will be much more noticeable. Also, there doesn't even appear to be support for the microbit in libtock rs https://github.com/tock/libtock-rs/tree/master/boards.

brghena, 2021-08-06:
I'm going to go with the minimum size of 4096 for now, however, we can and should revisit the question once small apps become a Tock reality. Maybe by then openocd will support smaller erase sizes on the nRF52s. But either way it will be a broader Tockloader question at that point.

(I'm also willing to change this decision if @alexandruradovici disagrees)

PR #72 (2021-08-09): Minimum size apps of 4096 as default for the microbit
Adds `minimum_size` of 4096 as a default command for the Microbit v2.

Tested on a Microbit v2 and resolves #67, which discusses tradeoffs.

Note: adding `bundle_apps' also resolves the problem.

_Edited: original version used `bundle_apps`_
closed 2021-08-09 by bradjc