tock/tockloader #68: Machine readable output

bradjc, 2021-04-23:
When using tockloader in a testing context, where a script is using tockloader to program a board, the current human-readable tockloader output is not particularly useful (and is also not guaranteed to change, and, perhaps relatedly, changes all of the time). Tockloader should support some sort of flag to give machine readable output.

Features to add this for:

- [ ] `tockloader list`: At the very least, getting the status of the board would be helpful. That way a tool can try to program the board, and then at least check the board is in the state it expects.

ppannuto, 2021-04-23:
@goodoomoodoo @jkchien this may be useful for your CI work. Right now, it's probably fine to simply verify the exit code of tockloader (i.e. did it just work), but it might be useful to think about what kinds of checks you'd hope to automate via CI that tockloader could help support.

bradjc, 2022-09-01:
This support now exists. Use flag `--output-format json` to get list, info, and list-attributes in JSON format.

Support is pretty basic, but I think there is enough scaffolding to build on it in the future if desired.

closed 2022-09-01 by bradjc