tock/tockloader #77: flashing with fails with v1.8.0

shenki, 2021-09-20:
I'm running tockloader installed with pip on my Debian system with an Arty A7 100T.

```
$ openocd --version
Open On-Chip Debugger 0.11.0
$ tockloader --version
1.8.0
$ tockloader install --openocd --board arty blink --debug
```

It fails to load the flash proxy, as it's looking inside the local python install path:

```
Debug: 158 32 command.c:146 script_debug(): command - pld load 0 /home/joel/.local/lib/python3.9/site-packages/tockloader/../bitfiles/bscan_spi_xc7a100t.bit
Error: 160 32 xilinx_bit.c:81 xilinx_read_bit_file(): couldn't stat() /home/joel/.local/lib/python3.9/site-packages/tockloader/../bitfiles/bscan_spi_xc7a100t.bit: No such file or directory
```

make install worked fine, as it uses the proxy from tock/boards/arty_e21/openocd.

The full output can be seen here:

http://paste.debian.net/hidden/1ecbb370/

bradjc, 2021-09-20:
Hmm, interesting. Nothing around that should have changed...

What do the contents look like of the `tockloader` folder? `ls -R /home/joel/.local/lib/python3.9/site-packages/tockloader`?

Tockloader expects there to be another folder there, like: `/usr/local/lib/python3.9/site-packages/tockloader-1.8.0.dev0-py3.9.egg/bitfields`

watsonkm, 2022-11-20:
My apologies, I ran into this issue and created a duplicate issue on accident (#95)! I think I've figured out the core issue, which I presented a possible fix for in the associated pull request.  Hopefully that should help!

watsonkm, 2022-11-20:
I think I had the same issue in #96, apologies for the duplicate issue! The pull request referenced there should (fingers crossed) fix this problem.

hudson-ayers, 2022-11-22:
Fixed by #96

closed 2022-11-22 by hudson-ayers