tock/tockloader #105: Failure when parsing a TBF with trailing padding.

jrvanwhy, 2023-11-01:
After updating `elf2tab` to a commit that includes https://github.com/tock/elf2tab/pull/81, I'm seeing the following crash in Tockloader:

```
[INFO   ] Using settings from KNOWN_BOARDS["opentitan_earlgrey"]
[INFO   ] Operating on flash file "bazel-out/k8-fastbuild-ST-2cc462681f62/bin/sw/device/silicon_owner/tock/tests/basic/image1.bin".
[INFO   ] Limiting flash size to 0x100000 bytes.
[STATUS ] Installing app on the board...
[DEBUG  ] start: Enter bootloader mode
[DEBUG  ] start: Determine current board
[DEBUG  ] start: Update board specific options
[DEBUG  ] Could not determine kernel version.
Traceback (most recent call last):
  File "/home/jrvanwhy/.cache/bazel/_bazel_jrvanwhy/33a7c06cd67612f3e3f6059a552c3123/execroot/lowrisc_opentitan/bazel-out/k8-opt-exec-2B5CBBC6-ST-e846b08c7501/bin/external/tockloader_deps/pypi__tockloader/rules_python_wheel_entry_point_tockloader.runfiles/tockloader_deps/pypi__tockloader/rules_python_wheel_entry_point_tockloader.py", line 5, in <module>
    rc = main()
  File "/home/jrvanwhy/.cache/bazel/_bazel_jrvanwhy/33a7c06cd67612f3e3f6059a552c3123/external/tockloader_deps/pypi__tockloader/tockloader/main.py", line 1459, in main
    args.func(args)
  File "/home/jrvanwhy/.cache/bazel/_bazel_jrvanwhy/33a7c06cd67612f3e3f6059a552c3123/external/tockloader_deps/pypi__tockloader/tockloader/main.py", line 168, in command_install
    tock_loader.install(tabs, replace=replace, erase=args.erase, sticky=args.sticky)
  File "/home/jrvanwhy/.cache/bazel/_bazel_jrvanwhy/33a7c06cd67612f3e3f6059a552c3123/external/tockloader_deps/pypi__tockloader/tockloader/tockloader.py", line 340, in install
    replacement_apps = self._extract_apps_from_tabs(tabs, arch)
  File "/home/jrvanwhy/.cache/bazel/_bazel_jrvanwhy/33a7c06cd67612f3e3f6059a552c3123/external/tockloader_deps/pypi__tockloader/tockloader/tockloader.py", line 1647, in _extract_apps_from_tabs
    app = tab.extract_app(arch)
  File "/home/jrvanwhy/.cache/bazel/_bazel_jrvanwhy/33a7c06cd67612f3e3f6059a552c3123/external/tockloader_deps/pypi__tockloader/tockloader/tab.py", line 85, in extract_app
    tabtbf = self._extract_tbf_from_filebuffer(tbf_filename, binary)
  File "/home/jrvanwhy/.cache/bazel/_bazel_jrvanwhy/33a7c06cd67612f3e3f6059a552c3123/external/tockloader_deps/pypi__tockloader/tockloader/tab.py", line 255, in _extract_tbf_from_filebuffer
    tbff = TBFFooter(tbfh, app_binary, binary[start_of_footers:])
  File "/home/jrvanwhy/.cache/bazel/_bazel_jrvanwhy/33a7c06cd67612f3e3f6059a552c3123/external/tockloader_deps/pypi__tockloader/tockloader/tbfh.py", line 1843, in __init__
    base = struct.unpack("<HH", buffer[0:4])
struct.error: unpack requires a buffer of 4 bytes
```

bradjc, 2023-11-01:
Oof. Can you run elf2tab with -v and send the output?

I wonder if the footer is 3 bytes (or 2 or 1) to make the size a multiple of 4.

jrvanwhy, 2023-11-01:
`elf2tab` with `-v` will be a bit of work (this is buried in OpenTitan's build system), but I'll do that if I need help.

The footer is 3 bytes, and it's all padding. It's a little unclear to me from the TBF docs how a TBF parser is supposed to distinguish the footer from padding. I'm guessing you're supposed to tell them apart by check if they're all zeros. Of course, anything less than 4 bytes can't be a footer.

bradjc, 2023-11-01:
Ok if you know that then it's an easy patch to tockloader. No need to run elf2tab.

jrvanwhy, 2023-11-01:
It looks to me like Tockloader is parsing the padding as a series of TLV entries with ID and length 0. Does that sound right to you?

jrvanwhy, 2023-11-02:
Fixed by #106

closed 2023-11-02 by jrvanwhy