tock/tockloader #125: Tockloader and Tockloader-rs produce different outputs

Joan-DwD, 2025-09-18:
I noticed a discrepancy between the Rust and Python Tockloade-rs where, for the init_fn_offset value, the Rust version (tockloader-rs/src/tbf_parser/types.rs at line 989) adds the full header size to the offset while the Python version (tockloader/tbfh.py at line 74) does not, and a similar issue exists for the protected_size value in the same files (Rust at line 959 and Python at line 75), which together cause a mismatch in the outputs between the two tools.

lschuermann, 2025-09-21:
Hey @Joan-DwD, thanks for raising this issue! At this point, `tockloader-rs` is not widely used across the core Tock maintainers yet, so `tockloader` is the canonical tool we'd recommend users to use. But this discrepancy is a good opportunity for us to cross-check both implementations, and what our documentation says about these header fields.

Tagging @bradjc and @george-cosma as the most active maintainers of both `tockloader` and `tockloader-rs` respectively.

george-cosma, 2025-09-21:
On `tockloader-rs` we use a modified version of `tock-tbf`. 

Just to leave a link to the relevant line here:

https://github.com/tock/tock/blob/72f743f88ce7081aa967e194ee391eddeb49e737/libraries/tock-tbf/src/types.rs#L762

In `tockloader` i also noticed quite an interesting comment. It suggests that the documentation implies something while the kernel uses it otherwise:

https://github.com/tock/tockloader/blob/7774669c41262486475bce3e7957e22c402dc807/tockloader/tbfh.py#L1170

lschuermann, 2025-09-22:
I knew this sounded familiar: at least for the `protected_region`, we fixed the documentation to match the behavior that is actually implemented in the kernel: https://github.com/tock/tock/pull/3515

bradjc, 2025-09-22:
Yes this was not very well specified initially, and different tools ended up doing different things. The language in PR 3515 is what we settled on and is the correct behavior.